TrustedFirmware-A/trusted-firmware-a #5: MISRA C issue - MISRA R15.7 All if...else if constructs shall be terminated with an else statement.

rajathpd, 2024-02-19:
Hello,

In the file /lib/el3_runtime/aarch64/context_mgmt.c. getting misra_c_2012_rule_15_7_violation:	No non-empty terminating "else" statement. during the Coverity MISRA-C analysis.

void cm_prepare_el3_exit(uint32_t security_state)
{
    u_register_t sctlr_elx, scr_el3, mdcr_el2;

if (security_state == NON_SECURE) {
    	scr_el3 = read_ctx_reg(get_el3state_ctx(ctx),
    					 CTX_SCR_EL3);
misra_c_2012_rule_15_7_violation:	No non-empty terminating "else" statement.
    	if ((scr_el3 & SCR_HCE_BIT) != 0U) {
    		...
    		...
    		...
    		write_sctlr_el2(sctlr_elx);
    	} else if (el_implemented(2) != EL_IMPL_NONE) {
    		el2_unused = true;
    		...
    		...
    		...
    		write_cnthp_ctl_el2(CNTHP_CTL_RESET_VAL &
    					~(CNTHP_CTL_ENABLE_BIT));
    	}
    	manage_extensions_nonsecure(el2_unused, ctx);
    }
cm_el1_sysregs_context_restore(security_state);
    cm_set_next_eret_context(security_state);
}

Resolution: added empty else statement to resolve the issue.

diff --git a/lib/el3_runtime/aarch64/context_mgmt.c b/lib/el3_runtime/aarch64/context_mgmt.c
index 8e6bfc5..8996746 100644
--- a/lib/el3_runtime/aarch64/context_mgmt.c
+++ b/lib/el3_runtime/aarch64/context_mgmt.c
@@ -789,6 +789,8 @@ void cm_prepare_el3_exit(uint32_t security_state)
    		 */
    		write_cnthp_ctl_el2(CNTHP_CTL_RESET_VAL &
    					~(CNTHP_CTL_ENABLE_BIT));
+		} else {
+			/* To fix the MISRA 15.7 warning - Added an empty else statement */
    	}
    	manage_extensions_nonsecure(el2_unused, ctx);

    }

-- 
Is it possible to add an empty else statement in this code? Please suggest.
[image]

manish-pandey-arm, 2024-03-06:
This patch would fix the issue https://review.trustedfirmware.org/c/TF-A/trusted-firmware-a/+/27042

odeprez, 2024-04-15:
Closing as patch https://review.trustedfirmware.org/c/TF-A/trusted-firmware-a/+/27042

closed 2024-04-15 by odeprez