TrustedFirmware-A/trusted-firmware-a #6: Error while building Arm clang tool chain

rajathpd, 2024-02-20:
Hello,

We have followed these steps below to build arm clang tool chain:

-export ARMLMD_LICENSE_FILE=8224@xsjlicsrvip
-export ARM_TOOL_VARIANT=ult
-export PATH=/tools/installs/arm/safety/armcc/6.6.2/bin:$PATH
-export ARM_INCLUDE=/tools/installs/arm/safety/armcc/6.6.2/include
-export XTOOLS_VER=2023
-export CROSS_COMPILE=/proj/xbuilds/2023.1_daily_latest/installs/lin64/Vitis/2023.1/gnu/aarch64/lin/aarch64---linux/bin/aarch64-linux-gnu-
-export PATH=/tools/installs/arm/safety/armcc/6.6.2/bin:$PATH
-export ARM_INCLUDE=/tools/installs/arm/safety/armcc/6.6.2/include
-make CC=/tools/installs/arm/safety/armcc/6.6.2/bin/armclang distclean
-make -j4 CC=/tools/installs/arm/safety/armcc/6.6.2/bin/armclang RESET_TO_BL31=1 PLAT=zynqmp bl31

1) For this, we get an error in mbranch-protection flag in plat/xilinx/zynqmp/platform.mk saying: unknown argument: 'mbranch-protection=none'. I've attached a snippet for the same.
[image]

Resolution:
In order to solve the above error, we commented -> TF_CFLAGS_aarch64  +=  -mbranch-protection=none
By doing so we could resolve this error.
 Is it a right way to solve the error?

2)  We get another error in include/arch/aarch64/smccc_helpers.h:114:23: error: expected expression
       smc_args_t ret_args ={0}; 
I've attached a snippet for the same.
[image]

Resolution: 
In order to solve this error, we removed the initialization:
      smc_args_t ret_args ={ }; 
Is it a right way to solve the error?

3) We get another error in lib/compiler-rt/builtins/int_lib.h :70:2: error: Unsupported target
#error Unsupported target
[image]

Resolution:
To solve the error, we defined: defined(__clang__)  in int_lib.h
Is it a right way to solve the error? Please suggest.

Thanks & regards,
Rajath P D

CJKay, 2024-03-06:
Hi @rajathpd, could you try with the latest version of the integration branch? We have merged some changes that I suspect might already resolve this.

rajathpd, 2024-04-04:
Hi @CJKay ,

I tried building the armclang by checking out to the integration branch but still got the same error. Please find the attached snippet of the error.
[image]

CJKay, 2024-04-05:
What version of Arm Clang are you using?

odeprez, 2024-05-27:
Hi is there any further update to this ticket? If not, I'm about to close it.

odeprez, 2024-07-09:
Closing as no update since Apr 24.

closed 2024-07-09 by odeprez