TrustedFirmware-A/trusted-firmware-a #7: Build failure under yocto

reatmon-ti, 2024-04-18:
We do nightly testing of our yocto layers against the latest kernel, uboot, trusted-firmware-a, and optee.  On April 12th we started getting a build failure with trusted-firmware-a.  I have tracked the issue down to this commit:

https://github.com/ARM-software/arm-trusted-firmware/commit/71c42e98bbe900ae806dfe87d2429d3622551789

Specifically, the line in make_helpers/utilities.mk:

escape-shell = '$(subst ','\'',$(1))'

On the surface it feels like the ' is overused and might cause issues.
I tried making the line:

escape-shell = $(subst ','\'',$(1))

And the builds went back to working properly.  Does that seem like a change that should be made, or was there a reason for the extra '' wrapper?

reatmon-ti, 2024-04-18:
Yes.  The Yocto recipes do that.  Here is an example from their generated build script:

export CC="aarch64-poky-linux-gcc  -mbranch-protection=standard -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/scratch/builds/poky-master-next/build/build/poky-tmp/work/am62xx_evm-poky-linux/trusted-firmware-a/2.10+git/recipe-sysroot"

So this change is going to break all of the Yocto builds.

CJKay, 2024-04-18:
Sorry to throw a spanner in the works!

TF-A doesn't expect flags to be passed as part of the toolchain variables. Can you instead pass these flags via `CFLAGS`? e.g.

```
export CC="aarch64-poky-linux-gcc"
export CFLAGS="-Wformat-security --sysroot=/scratch/builds/poky-master-next/build/build/poky-tmp/work/am62xx_evm-poky-linux/trusted-firmware-a/2.10+git/recipe-sysroot"
```

The build system also expects to control some of the flags that are being passed here, so passing them even via `CFLAGS` is unlikely to produce the desired result:

- `-O2`: TF-A always builds with `-Os`. Passing optimisation flags might work for now but is liable to change at any moment, and we can't make any guarantee that this flag will apply to all or even some of the sources currently. See: https://github.com/TrustedFirmware-A/trusted-firmware-a/blob/c8be7c08c3b3a2330d54b58651faa9438ff34f6e/Makefile#L288-L291
- `-mbranch-protection=standard`: branch protection is controlled by the `BRANCH_PROTECTION` build option as it needs software-level support; for `-mbranch-protection=1` specifically that's `BRANCH_PROTECTION=1. See: https://github.com/TrustedFirmware-A/trusted-firmware-a/blob/main/Makefile#L509-L537
- `-fstack-protector-strong`: stack protection also needs software-level support, and should be enabled by the `ENABLE_STACK_PROTECTOR=1` build option. See: https://github.com/TrustedFirmware-A/trusted-firmware-a/blob/main/lib/stack_protector/stack_protector.mk

reatmon-ti, 2024-04-18:
This is not me.  This is the Yocto project.  This is how it is structured for all packages.  But if you wan to try and fix it, then meta-arm is the layer where the recipe is.  And it is maintained by you guys.

rossburton, 2024-04-18:
Yes, this is an issue in meta-arm, not TF-A.

We'll look at it, I suspect this is quite simply and we've been doing the wrong thing for a while but getting away with it.

glneo, 2024-04-18:
This is also breaking Armbian(Debian) builds:
```
/bin/sh: 1: ccache aarch64-linux-gnu-gcc: not found
make: ccache aarch64-linux-gnu-gcc: No such file or directory
```
Seems `CC="ccache gcc"` will now be broken.

CJKay, 2024-04-18:
> This is also breaking Armbian(Debian) builds:
> 
> ```
> /bin/sh: 1: ccache aarch64-linux-gnu-gcc: not found
> make: ccache aarch64-linux-gnu-gcc: No such file or directory
> ```
> 
> Seems `CC="ccache gcc"` will now be broken.

Hi, @glneo, ccache has a second run mode intended for this sort of situation. See bullet 2 from: https://github.com/ccache/ccache/blob/master/doc/MANUAL.adoc#run-modes. It's my understanding from talking to others using ccache that this works.

If this is too inconvenient then we could consider a build option along the lines of what CMake provides ([CMAKE_<LANG>_COMPILER_LAUNCHER](https://cmake.org/cmake/help/latest/envvar/CMAKE_LANG_COMPILER_LAUNCHER.html)).

reatmon-ti, 2024-04-19:
There seems to be a lot of breakage around existing build systems and how they interact with the CC variable, and the course taken seems to be "Fix all of the other build systems to help the Windows users.".

An alternative idea would be to not break existing systems and introduce a new WINDOWS_CC variable (or something like it) to hold the path to the compiler on a Windows host and then do the escape-shell on that variable and then stick the result in CC in the Makefile.  And only do all of that on a Windows host.

CJKay, 2024-05-16:
FYI [this patch](https://review.trustedfirmware.org/c/TF-A/trusted-firmware-a/+/28839) aiming for v2.11-rc1 should resolve this issue.

odeprez, 2024-05-27:
Hi assuming Chris' change now merged fixes on going issues, I'm going to close this ticket if everyone agrees.

odeprez, 2024-07-09:
Closing as considered completed and no update since May 24.

closed 2024-07-09 by odeprez