TrustedFirmware-A/trusted-firmware-a #24: TF-A failing to build on aarch64 machine

eballetbo, 2024-08-02:
While cross compiling TF-A works without problems the toolchain helpers are not able to properly provide a toolchain set for native compilation or building on a aarch64 machine. It fails with the following error as is looking for a cross toolchain.

```
make PLAT=k3 TARGET_BOARD=j784s4 SPD=opteed K3_USART=0x8 bl31
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The configured AArch64 C compiler could not be identified and may not be supported:
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344:     aarch64-none-elf-gcc
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The default AArch64 C compiler is:
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344:     aarch64-none-elf-gcc
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The following tools are supported:
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344:   - Arm® Compiler for Embedded `armclang`
make_helpers/toolchain.mk:344:   - LLVM Clang (`clang`)
make_helpers/toolchain.mk:344:   - GNU GCC (`gcc`)
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The build system will treat this C compiler as GNU GCC (`gcc`).
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The configured AArch64 C preprocessor could not be identified and may not be supported:
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344:     aarch64-none-elf-gcc
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The default AArch64 C preprocessor is:
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344:     aarch64-none-elf-gcc
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The following tools are supported:
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344:   - Arm® Compiler for Embedded `armclang`
make_helpers/toolchain.mk:344:   - LLVM Clang (`clang`)
make_helpers/toolchain.mk:344:   - GNU GCC (`gcc`)
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The build system will treat this C preprocessor as GNU GCC (`gcc`).
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The configured AArch64 assembler could not be identified and may not be supported:
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344:     aarch64-none-elf-gcc
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The default AArch64 assembler is:
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344:     aarch64-none-elf-gcc
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The following tools are supported:
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344:   - Arm® Compiler for Embedded `armclang`
make_helpers/toolchain.mk:344:   - LLVM Clang (`clang`)
make_helpers/toolchain.mk:344:   - GNU GCC (`gcc`)
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The build system will treat this assembler as GNU GCC (`gcc`).
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The configured AArch64 linker could not be identified and may not be supported:
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344:     aarch64-none-elf-gcc
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The default AArch64 linker is:
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344:     aarch64-none-elf-gcc
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The following tools are supported:
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344:   - Arm® Compiler for Embedded `armclang`
make_helpers/toolchain.mk:344:   - Arm® Compiler for Embedded `armlink`
make_helpers/toolchain.mk:344:   - LLVM Clang (`clang`)
make_helpers/toolchain.mk:344:   - LLVM LLD (`lld`)
make_helpers/toolchain.mk:344:   - GNU GCC (`gcc`)
make_helpers/toolchain.mk:344:   - GNU LD (`ld.bfd`)
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The build system will treat this linker as GNU GCC (`gcc`).
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The configured AArch64 object copier could not be identified and may not be supported:
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344:     aarch64-none-elf-objcopy
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The default AArch64 object copier is:
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344:     aarch64-none-elf-objcopy
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The following tools are supported:
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344:   - Arm® Compiler for Embedded `fromelf`
make_helpers/toolchain.mk:344:   - LLVM `llvm-objcopy`
make_helpers/toolchain.mk:344:   - GNU `objcopy`
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The build system will treat this object copier as GNU `objcopy`.
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The configured AArch64 object dumper could not be identified and may not be supported:
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344:     aarch64-none-elf-objdump
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The default AArch64 object dumper is:
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344:     aarch64-none-elf-objdump
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The following tools are supported:
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344:   - Arm® Compiler for Embedded `fromelf`
make_helpers/toolchain.mk:344:   - LLVM `llvm-objdump`
make_helpers/toolchain.mk:344:   - GNU `objdump`
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The build system will treat this object dumper as GNU `objdump`.
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The configured AArch64 archiver could not be identified and may not be supported:
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344:     aarch64-none-elf-gcc-ar
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The default AArch64 archiver is:
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344:     aarch64-none-elf-gcc-ar
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The following tools are supported:
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344:   - Arm® Compiler for Embedded `armar`
make_helpers/toolchain.mk:344:   - LLVM `llvm-ar`
make_helpers/toolchain.mk:344:   - GNU `ar`
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The build system will treat this archiver as GNU `ar`.
make_helpers/toolchain.mk:344: 
Including services/spd/opteed/opteed.mk
  MD      /home/fedora/sources/trusted-firmware-a/build/k3/j784s4/release/bl31
  CC      bl31/bl31_context_mgmt.c
make: aarch64-none-elf-gcc: No such file or directory
make: *** [Makefile:1516: /home/fedora/sources/trusted-firmware-a/build/k3/j784s4/release/bl31/bl31_context_mgmt.o] Error 127

```

eballetbo, 2024-08-02:
Actually I think you can pass `CC=gcc CPP=gcc AS=gcc LD=gcc OC=objcopy AR=gcc-ar` to solve this, but would be good have native building on aarch64 just working.

nullr0ute, 2024-08-05:
This wasn't an issue with 2.10.x has anyone tried to bisect it to see which commit broke it?

eballetbo, 2024-08-05:
I did a quick bisect run and pointed out to this commit. I didn't look further yet.

```
commit ffb7742125def3e0acca4c7e4d3215af5ce25a31 (HEAD)
Author: Chris Kay <chris.kay@arm.com>
Date:   Mon Dec 4 12:03:51 2023 +0000

    build: use new toolchain variables for tools
    
    This change migrates the values of `CC`, `CPP`, `AS` and other toolchain
    variables to the new `$(toolchain)-$(tool)` variables, which were
    introduced by the toolchain refactor patch. These variables should be
    equivalent to the values that they're replacing.
    
    Change-Id: I644fe4ce82ef1894bed129ddb4b6ab94fb04985d
    Signed-off-by: Chris Kay <chris.kay@arm.com>
```

odeprez, 2024-08-26:
Cc @CJKay 

Hi this problem was discussed in https://lists.trustedfirmware.org/archives/list/tf-a@lists.trustedfirmware.org/thread/RV7Y4V3NBIDCFIV65UYXD4XTMNNV5QJP/

As mentioned above and in the ML , the usage is about providing CC variable. Would that be enough to fix build using TF-A v2.11 onwards? Or do we expect additional changes?

CJKay, 2024-08-28:
Hi @eballetbo, @nullr0ute, sorry - I didn't see this. It should be sufficient to just pass `CC=gcc` - the rest of the toolchain will be picked up automatically, so you don't need to provide `CPP`, `AS`, etc.

odeprez, 2024-08-30:
Closing following resolution proposed by Chris.

closed 2024-08-30 by odeprez

nullr0ute, 2024-09-04:
So it doesn't appear to work as suggested, I get the following if I just set `CC=gcc` when building:

`+ make 'HOSTCC=gcc -O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Wno-complain-wrong-lang -Werror=format-security -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -mbranch-protection=standard -fasynchronous-unwind-tables -fstack-clash-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer' CROSS_COMPILE= PLAT=axg bl31
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The configured AArch64 archiver could not be identified and may not be supported:
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344:     aarch64-none-elf-gcc-ar
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The default AArch64 archiver is:
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344:     aarch64-none-elf-gcc-ar
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The following tools are supported:
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344:   - Arm® Compiler for Embedded `armar`
make_helpers/toolchain.mk:344:   - LLVM `llvm-ar`
make_helpers/toolchain.mk:344:   - GNU `ar`
make_helpers/toolchain.mk:344: 
make_helpers/toolchain.mk:344: The build system will treat this archiver as GNU `ar`.
make_helpers/toolchain.mk:344: `

And then when it tries to use ar:

`  AR      /builddir/build/BUILD/arm-trusted-firmware-2.11.0-build/trusted-firmware-a-2.11.0/build/axg/release/lib/libc.a
make: aarch64-none-elf-gcc-ar: No such file or directory
make: *** [Makefile:1491: /builddir/build/BUILD/arm-trusted-firmware-2.11.0-build/trusted-firmware-a-2.11.0/build/axg/release/lib/libc.a] Error 127`

CJKay, 2024-09-04:
@nullr0ute Could you post the results of:

```
which aarch64-linux-gnu-gcc
which aarch64-linux-gnu-gcc-ar
```

odeprez, 2024-09-06:
Submitted Chris' change. Let us know if this helps, thanks.

nullr0ute, 2024-09-06:
Yes, it worked for me, thanks!