TrustedFirmware-A/trusted-firmware-a #26: Build failure of TF-A v2.11 under Buildroot

vstehle, 2024-09-02:
Switching from TF-A v2.10 to v2.11 can cause build issues under [Buildroot](https://buildroot.org/).

In some cases, the link fails with the following error:

```
bl1.elf: error: PHDR segment not covered by LOAD segment
```

This can be reproduced with the following commands:

```
git clone https://gitlab.com/buildroot.org/buildroot.git -b 2024.08-rc3
cd buildroot
make qemu_aarch64_ebbr_defconfig
sed -i 's/v2.10/v2.11/' .config
make arm-trusted-firmware
```

(To confirm that v2.10 is indeed building fine, just omit the line with the `sed` command. The file `output/images/bl1.bin` should be produced.)

The issue seems to be caused by the switch from `ld` to `gcc` for the link of TF-A bl1, combined with the fact that Buildroot uses a toolchain wrapper with PIE related flags.

jolivain, 2024-09-02:
Hi @vstehle,

I'm currently able to fix this issue with the following Buildroot TF-A build recipe patch:
https://github.com/jolivain/buildroot/commit/fa3fd91abf444fe070708d4fd8123c3f9ad5c774
See the commit log for my current analysis.

I believe this patch will be needed in Buildroot anyways because it needs to compile many TF-A different versions, ranging from v2.0 to 2.10 and hopefully v2.11 soon. So even if there is an improvement in TF-A, it will appear in newer versions only. I need to run some more tests to check if my patch is not breaking anything (in older TF-A version) before sending it.

Detecting PIE just by looking at the "gcc -v" output might have some limitation. See here:
https://git.trustedfirmware.org/plugins/gitiles/TF-A/trusted-firmware-a.git/+/refs/tags/v2.11.0/Makefile#711

Maybe one way to improve could be to enhance the "PIE_FOUND" detection (by compiling a small test program) or to provide a variable to declare that the toolchain will do PIE (e.g. TOOLCHAIN_GENERATES_PIE=1) that will override the PIE_FOUND detection?

vstehle, 2024-09-03:
Thanks @jolivain; I confirm your patch allows to build v2.11 with Buildroot.

jolivain, 2024-09-03:
Hi @vstehle, thanks testing!  I'll add a `Tested-by:` tag when I'll submit my patch.  I'm still running regression testing on all defconfigs that include a version of TF-A...  Hopefully, I'll send the patch later today.

jolivain, 2024-09-03:
For info, I proposed a patch in Buildroot here:
https://patchwork.ozlabs.org/project/buildroot/patch/20240903180529.220109-1-ju.o@free.fr/

odeprez, 2024-09-06:
Hi @jolivain thanks for digging and fixing into buildroot build system. What is the way forward? Do you expect further refinement on TF-A build flow side?

odeprez, 2025-05-05:
Closing as the conversation has stalled in last Oct'25. Hopefully this issue related to older TF-A v2.11 is fixed by patches in other projects (e.g. buildroot). Thanks.

closed 2025-05-05 by odeprez