TrustedFirmware-A/trusted-firmware-a #27: imx8mq: Fail to build due to RAM overflow

fabioestevam, 2024-09-10:
Since commit dd108c3c1fe3f958a38ae255e57b41e5453d077f

```
commit dd108c3c1fe3f958a38ae255e57b41e5453d077f
Author: Jacky Bai <ping.bai@nxp.com>
Date:   Tue Jan 7 16:44:46 2020 +0800

    feat(imx8mq): add the dram retention support for imx8mq
    
    Add the dram retention support for i.MX8MQ. As there is
    no enough ocram space available before entering TF-A,
    so the timing info need to be copied from dram into ocram.
    
    Signed-off-by: Jacky Bai <ping.bai@nxp.com>
    Change-Id: Id8264c342fd62e297b1969cba5ed505450c78a25
```

It is no longer possible to build TF-A for imx8mq because the internal RAM size overflowed:

```
make PLAT=imx8mq bl31
...
  AR      /home/fabio/trusted-firmware-a/build/imx8mq/release/lib/libc.a
  LD      /home/fabio/trusted-firmware-a/build/imx8mq/release/bl31/bl31.elf
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: /home/fabio/trusted-firmware-a/build/imx8mq/release/bl31/bl31.elf section `.xlat_table' will not fit in region `RAM'
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: BL31 image has exceeded its limit.
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: region `RAM' overflowed by 20480 bytes
collect2: error: ld returned 1 exit status
make: *** [Makefile:1524: /home/fabio/trusted-firmware-a/build/imx8mq/release/bl31/bl31.elf] Error 1
```
@JackyBai please take a look.

v2.8 builds fine on imx8mq. v2.9 to v2.11 fails to build.

odeprez, 2024-09-12:
Ping other NXP platform maintainers @MrVan @pangupta
@madhukar-Arm (Arm platform liaison)

madhukar-Arm, 2024-09-12:
Based on this patch from @JackyBai , this platform build is broken from a long time.
https://review.trustedfirmware.org/c/ci/tf-a-ci-scripts/+/10229

fabioestevam, 2024-09-12:
@madhukar-Arm @JackyBai 

Doesn't NXP have an interest in maintaining i.MX8MQ TF-A support any longer?

If so, please fix it.

Otherwise, remove i.MX8MQ support.

JackyBai, 2024-09-18:
Hi Fabio,

i.MX8MQ still need to be supported, but due to the too small OCRAM limitation, it can not be build successfully with mainline TF-A for now. if we want to make it build with some features dropped, I can workout a patch to fix it. 

For this SoC, suggested to use the NXP release as in NXP yocto release, we add some special hack to make it work with no feature dropped.

odeprez, 2024-09-18:
I gave a quick try and it seems the platform has deviated a bit more than just the RAM issue:

> make CROSS_COMPILE=aarch64-none-elf- PLAT=imx8mq 
> 
> plat/imx/common/imx_sip_svc.c: In function 'imx_sip_handler':
> plat/imx/common/imx_sip_svc.c:86:34: error: implicit declaration of function 'imx_hab_handler'; did you mean 'imx_sip_handler'? [-Werror=implicit-function-declaration]
>    86 |                 SMC_RET1(handle, imx_hab_handler(smc_fid, x1, x2, x3, x4));
>       |                                  ^~~~~~~~~~~~~~~
> include/lib/el3_runtime/aarch64/context.h:237:56: note: in definition of macro 'write_ctx_reg'
>   237 |                                          = (uint64_t) (val))
>       |                                                        ^~~
> plat/imx/common/imx_sip_svc.c:86:17: note: in expansion of macro 'SMC_RET1'
>    86 |                 SMC_RET1(handle, imx_hab_handler(smc_fid, x1, x2, x3, x4));
>       |                 ^~~~~~~~
> cc1: all warnings being treated as errors
> 

Beyond the above, would it be possible to pass platform options such that the platform can at least build in the CI even if those options do not reflect a real world usage?

fabioestevam, 2024-09-18:
Hi Jacky,

> Hi Fabio,
> 
> i.MX8MQ still need to be supported, but due to the too small OCRAM limitation, it can not be build successfully with mainline TF-A for now. if we want to make it build with some features dropped, I can workout a patch to fix it.

Please submit a patch so TF-A mainline can build and work on i.MX8MQ.
 
> For this SoC, suggested to use the NXP release as in NXP yocto release, we add some special hack to make it work with no feature dropped.

The latest NXP TF-A also fails. 

Please see this report:
https://github.com/Freescale/meta-freescale/issues/1937

odeprez, 2025-01-06:
Hi @JackyBai 

>  if we want to make it build with some features dropped, I can workout a patch to fix it.

Just a gentle ping, do you still intend to produce a patch?

fabioestevam, 2025-02-27:
Hi @odeprez and @JackyBai 

How do you think we should proceed?

NXP is not interested in supporting imx8mq with mainline trusted-firmware-a.

I can submit a patch removing the imx8mq support.

madhukar-Arm, 2025-02-27:
I think we can initiate deprecation of the platform if it fails to build. The guidelines are provided here: https://trustedfirmware-a.readthedocs.io/en/latest/process/platform-ports-policy.html#deprecation-policy

odeprez, 2025-05-05:
Hi @JackyBai fyi we'll proceed with deprecation and removal of the platform unless a fix is provided for building. Please let us know if you disagree.