TrustedFirmware-A/trusted-firmware-a #37: [DebianTesting(Trixie)] make PLAT=rk3588 fails with binutils new warning

aarnoud, 2025-01-16:
```
# git rev-parse HEAD
f532cd306989041ca479204aa6c118421d1e5f51
```

```
# make PLAT=rk3588 all
  MD      /root/trusted-firmware-a/build/rk3588/release/libfdt
  CC      lib/libfdt/fdt.c
  CC      lib/libfdt/fdt_addresses.c
  CC      lib/libfdt/fdt_empty_tree.c
  CC      lib/libfdt/fdt_ro.c
  CC      lib/libfdt/fdt_rw.c
  CC      lib/libfdt/fdt_strerror.c
  CC      lib/libfdt/fdt_sw.c
  CC      lib/libfdt/fdt_wip.c
  MD      /root/trusted-firmware-a/build/rk3588/release/lib
  AR      /root/trusted-firmware-a/build/rk3588/release/lib/libfdt.a
Building rk3588
  MD      /root/trusted-firmware-a/build/rk3588/release/libc
  CC      lib/libc/abort.c
  CC      lib/libc/assert.c
  CC      lib/libc/exit.c
  CC      lib/libc/memchr.c
  CC      lib/libc/memcmp.c
  CC      lib/libc/memcpy.c
  CC      lib/libc/memcpy_s.c
  CC      lib/libc/memmove.c
  CC      lib/libc/memrchr.c
  CC      lib/libc/printf.c
  CC      lib/libc/putchar.c
  CC      lib/libc/puts.c
  CC      lib/libc/snprintf.c
  CC      lib/libc/strchr.c
  CC      lib/libc/strcmp.c
  CC      lib/libc/strlcat.c
  CC      lib/libc/strlcpy.c
  CC      lib/libc/strlen.c
  CC      lib/libc/strncmp.c
  CC      lib/libc/strnlen.c
  CC      lib/libc/strrchr.c
  CC      lib/libc/strtok.c
  CC      lib/libc/strtoul.c
  CC      lib/libc/strtoll.c
  CC      lib/libc/strtoull.c
  CC      lib/libc/strtol.c
  CC      lib/libc/memset.c
  AS      lib/libc/aarch64/setjmp.S
  AR      /root/trusted-firmware-a/build/rk3588/release/lib/libc.a
  MD      /root/trusted-firmware-a/build/rk3588/release/bl31
  CC      bl31/bl31_context_mgmt.c
  CC      bl31/bl31_main.c
  CC      bl31/bl31_traps.c
  CC      bl31/interrupt_mgmt.c
  CC      common/bl_common.c
  CC      common/desc_image_load.c
  CC      common/runtime_svc.c
  CC      common/tf_log.c
  CC      drivers/arm/gic/v3/arm_gicv3_common.c
  CC      drivers/arm/gic/v3/gic-x00.c
  CC      drivers/arm/gic/v3/gicdv3_helpers.c
  CC      drivers/arm/gic/v3/gicrv3_helpers.c
  CC      drivers/arm/gic/v3/gicv3_helpers.c
  CC      drivers/arm/gic/v3/gicv3_main.c
  CC      drivers/console/multi_console.c
  CC      drivers/delay_timer/delay_timer.c
  CC      drivers/delay_timer/generic_delay_timer.c
  CC      drivers/scmi-msg/base.c
  CC      drivers/scmi-msg/clock.c
  CC      drivers/scmi-msg/entry.c
  CC      drivers/scmi-msg/reset_domain.c
  CC      drivers/scmi-msg/smt.c
  CC      lib/bl_aux_params/bl_aux_params.c
  CC      lib/compiler-rt/builtins/popcountdi2.c
  CC      lib/compiler-rt/builtins/popcountsi2.c
  CC      lib/cpus/errata_common.c
  CC      lib/cpus/errata_report.c
  CC      lib/el3_runtime/aarch64/context_mgmt.c
  CC      lib/el3_runtime/cpu_data_array.c
  CC      lib/el3_runtime/simd_ctx.c
  CC      lib/extensions/mpam/mpam.c
  CC      lib/extensions/pmuv3/aarch64/pmuv3.c
  CC      lib/extensions/spe/spe.c
  CC      lib/extensions/sve/sve.c
  CC      lib/locks/bakery/bakery_lock_normal.c
  CC      lib/psci/psci_common.c
  CC      lib/psci/psci_main.c
  CC      lib/psci/psci_mem_protect.c
  CC      lib/psci/psci_off.c
  CC      lib/psci/psci_on.c
  CC      lib/psci/psci_setup.c
  CC      lib/psci/psci_suspend.c
  CC      lib/psci/psci_system_off.c
  CC      lib/xlat_tables_v2/aarch64/xlat_tables_arch.c
  CC      lib/xlat_tables_v2/xlat_tables_context.c
  CC      lib/xlat_tables_v2/xlat_tables_core.c
  CC      lib/xlat_tables_v2/xlat_tables_utils.c
  CC      plat/common/aarch64/plat_common.c
  CC      plat/common/plat_bl_common.c
  CC      plat/common/plat_gicv3.c
  CC      plat/common/plat_log_common.c
  CC      plat/common/plat_psci_common.c
  CC      plat/rockchip/common/aarch64/platform_common.c
  CC      plat/rockchip/common/bl31_plat_setup.c
  CC      plat/rockchip/common/params_setup.c
  CC      plat/rockchip/common/plat_pm.c
  CC      plat/rockchip/common/plat_pm_helpers.c
  CC      plat/rockchip/common/plat_topology.c
  CC      plat/rockchip/common/rockchip_gicv3.c
  CC      plat/rockchip/common/rockchip_sip_svc.c
  CC      plat/rockchip/common/scmi/scmi.c
  CC      plat/rockchip/common/scmi/scmi_clock.c
  CC      plat/rockchip/common/scmi/scmi_rstd.c
  CC      plat/rockchip/rk3588/drivers/pmu/pm_pd_regs.c
  CC      plat/rockchip/rk3588/drivers/pmu/pmu.c
  CC      plat/rockchip/rk3588/drivers/scmi/rk3588_clk.c
  CC      plat/rockchip/rk3588/drivers/scmi/rk3588_rstd.c
  CC      plat/rockchip/rk3588/drivers/secure/secure.c
  CC      plat/rockchip/rk3588/drivers/soc/soc.c
  CC      plat/rockchip/rk3588/plat_sip_calls.c
  CC      services/arm_arch_svc/arm_arch_svc_setup.c
  CC      services/std_svc/std_svc_setup.c
  AS      bl31/aarch64/bl31_entrypoint.S
  AS      bl31/aarch64/crash_reporting.S
plat/rockchip/common/include/plat_macros.S: Assembler messages:
plat/rockchip/common/include/plat_macros.S:15: Warning: entity size for SHF_MERGE / SHF_STRINGS not specified
plat/rockchip/common/include/plat_macros.S:33: Warning: entity size for SHF_MERGE / SHF_STRINGS not specified
/tmp/ccLuj7BF.s: Error: 2 warnings, treating warnings as errors
make: *** [Makefile:1587: /root/trusted-firmware-a/build/rk3588/release/bl31/crash_reporting.o] Error 1
```

CJKay, 2025-03-10:
Patch discussion [here](https://review.trustedfirmware.org/c/TF-A/trusted-firmware-a/+/34306) - we believe this is resolved in upstream binutils.

aarnoud, 2025-05-07:
It's building now, thanks

closed 2025-05-07 by aarnoud