TrustedFirmware-A/trusted-firmware-a #81: DLME data size comparison is mixing byte size and page size

stuyod01, 2025-02-10:
The check below compares dlme_data_max_size and dlme_data_min_size and seems to be mixing a comparison of the max count in bytes and min size in number of pages:

https://github.com/ARM-software/arm-trusted-firmware/blob/master/services/std_svc/drtm/drtm_main.c#L238

```
dlme_data_max_size = args->dlme_size - args->dlme_data_off;
if (dlme_data_max_size < dlme_data_min_size) {

                ERROR("%s: assertion failed:"

                      " dlme_data_max_size (%ld) < dlme_data_total_bytes_req (%ld)\n",

                      __func__, dlme_data_max_size, dlme_data_min_size);

                panic();

             }
```

However max size is a byte count value.  And min size is the number of pages:

https://github.com/ARM-software/arm-trusted-firmware/blob/master/services/std_svc/drtm/drtm_main.c#L115

```
dlme_data_min_size += dlme_data_hdr_init.dlme_addr_map_size +

                              ARM_DRTM_MIN_EVENT_LOG_SIZE +

                              dlme_data_hdr_init.dlme_tcb_hashes_table_size +

                              dlme_data_hdr_init.dlme_impdef_region_size;

    dlme_data_min_size = page_align(dlme_data_min_size, UP)/PAGE_SIZE;
```

ManishVB-Arm, 2025-02-11:
Hi Stuart,

Yes, this seems to be an issue. I provided a fix here: https://review.trustedfirmware.org/c/TF-A/trusted-firmware-a/+/35181 which is review now.

Thanks,
Manish Badarkhe

odeprez, 2025-02-28:
Fix is merged now, closing.

closed 2025-02-28 by odeprez