TrustedFirmware-A/trusted-firmware-a #109: build fail for TI k3

montjoie, 2025-03-14:
On top of master or lts-v2.12.0 I hit the following build failure:

make PLAT=k3 BUILD_BASE=/home/compile/crossbuild/ATF/arm64/ATF-beagleia64 ARCH=aarch64 SPD=opteed TARGET_BOARD=generic CROSS_COMPILE=aarch64-unknown-linux-gnu-
Including services/spd/opteed/opteed.mk
  CLEAN
Including services/spd/opteed/opteed.mk
Building k3
  MD      /home/compile/crossbuild/ATF/arm64/ATF-beagleia64/k3/generic/release/libc
  CC      lib/libc/abort.c
  CC      lib/libc/assert.c
  CC      lib/libc/exit.c
  CC      lib/libc/memchr.c
  CC      lib/libc/memcmp.c
[...]
  AS      services/spd/opteed/opteed_helpers.S
  MD      /home/compile/crossbuild/ATF/arm64/ATF-beagleia64/k3/generic/release/bl31/bl31
  PP      bl31/bl31.ld.S
  LD      /home/compile/crossbuild/ATF/arm64/ATF-beagleia64/k3/generic/release/bl31/bl31.elf
/usr/libexec/gcc/aarch64-unknown-linux-gnu/ld: warning: creating DT_TEXTREL in a PIE
collect2: error: ld returned 1 exit status
make: *** [Makefile:1556: /home/compile/crossbuild/ATF/arm64/ATF-beagleia64/k3/generic/release/bl31/bl31.elf] Error 1

Using gcc version 14.2.1 20241221 (Gentoo 14.2.1_p20241221 p7)

govindraj-arm, 2025-03-17:
Hi @montjoie ,

I see you're using gcc-14.2, I have not tried with that version but based on https://trustedfirmware-a.readthedocs.io/en/v2.12.0/getting_started/prerequisites.html

TF-A 2.12 was tested with 13.3 gcc and recommends the same.

Testing based on that build works.

```
trusted-firmware-a$ git log --oneline --decorate -1
4ec2948fe (HEAD -> intg, tag: v2.12.0, tag: v2.12, tag: lts-v2.12.0) Merge "docs(changelog): changelog for v2.12 release" into integration

trusted-firmware-a$ make PLAT=k3 ARCH=aarch64 SPD=opteed TARGET_BOARD=generic -j4 -s
Including services/spd/opteed/opteed.mk

trusted-firmware-a$ aarch64-none-elf-gcc --version
aarch64-none-elf-gcc (Arm GNU Toolchain 13.3.Rel1 (Build arm-13.24)) 13.3.1 20240614
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

```

odeprez, 2025-05-05:
Closing with the recommendation: LTS v2.10 shall be built with GCC 13. TI maintainers can provide downstream changes for bumping the toolchain from gcc 13 to 14 however tbd if this may be picked upstream.

closed 2025-05-05 by odeprez