TrustedFirmware-A/trusted-firmware-a #110: Fix CVE/erratum order in neoverse_v2.S

Idan-anpa, 2025-03-16:
Hi,
The following patch https://github.com/TrustedFirmware-A/trusted-firmware-a/commit/878464f02a043207f29282f05109012152df569b
Added a CVE before erratas.

This causes runtime errors in neoverse_v2 when running with FEATURE_DETECTION, as expected order is erratas (sorted by index), and then CVEs (sorted by year and then index)

CVE needs to be placed later in the file to conform with expected order

Thanks,
Idan

odeprez, 2025-03-17:
Hi @SonaMathew-arm @bipinravi-arm any idea how to fix the reported issue? Thanks.

SonaMathew-arm, 2025-03-19:
Hi Idan, 

Thank you for bringing this issue to our attention and for your valuable feedback. We have discussed the issue internally with our team and have decided on a 2 stage fix in terms of the changes. The first fix will be posted soon. 
The second stage of the fix is disassociating the ordering from FEATURE_DETECTION flag at runtime and changing it to a warning which we will also post shortly. 

Thanks
-Sona

SonaMathew-arm, 2025-03-21:
Hi Idan, 

The initial resolution for the above mentioned issue has been completed and merged. The patch stack can be accessed here https://review.trustedfirmware.org/q/topic:%22ar/cvereorder%22. 
The subsequent patch involving eliminating the warnings and implementing the errata/CVE ID ordering as part of a static check is still in progress and will be posted shortly too. 

Thanks!
-Sona

odeprez, 2025-05-05:
Closing as mentioned by Sona the patch stack addressing the issue initially reported is merged now.

closed 2025-05-05 by odeprez