TrustedFirmware-A/trusted-firmware-a #114: tegra: bl31 fails to initialize when compiled with clang

webgeek1234, 2025-04-14:
Building bl31 for a tegra arch like tegra186 using gcc results in a working binary that boots as expected. For example:
```
make CROSS_COMPILE=aarch64-linux-gnu- PLAT=tegra TARGET_SOC=t186 bl31
```

However, building the same target with clang results in a binary that fails to print anything to uart at all, even with full debug enabled. For example:
```
make CC=clang PLAT=tegra TARGET_SOC=t186 bl31
```

This has been a long standing issue. It was a problem on v2.6. I've most recently replicated on the v2.12.0 tag. I don't recall which clang version was in use when I first tried. The clang version I'm trying v2.12 against is 19.01.

odeprez, 2025-04-14:
Tagging @vwadekar TF-A Tegra platform maintainer.

vwadekar, 2025-04-22:
@odeprez we have not validated Tegra platforms with clang. Are we changing compilers for all platforms? Or is this a new activity?

Trying to understand the priority of this activity.

webgeek1234, 2025-04-22:
I can't speak for odeprez or project policy, but I can respond as the creator of the issue. This is a bug report, not a project activity.

https://github.com/TrustedFirmware-A/trusted-firmware-a/blob/main/docs/getting_started/initial-build.rst
The build docs list gcc, armclang, and clang as supported compilers.
https://github.com/TrustedFirmware-A/trusted-firmware-a/blob/main/docs/plat/nvidia-tegra.rst
The tegra docs do not say that clang is unsupported. So a reasonable reading of the docs put together is that tegra works when built with clang.

For a point of reference, my use case is building tf-a within the android/aosp build system. Google has deprecated use of gcc toolchains. While it is still possible to use custom gcc toolchains in build rules, it's becoming more difficult to do so. Hence using clang would be the path of least resistance. But at the very least, if clang is to be explicitly not supported for tegra, it would be nice to note that in the docs.

vwadekar, 2025-04-22:
Thanks for the context. Although we dont have immediate plans to support clang, we can try to evaluate this request internally and see if we can unblock you.

odeprez, 2025-04-23:
Thanks both for your answers, and acknowledging on the reasoning and route forward.

odeprez, 2025-05-05:
Hi @vwadekar 
Should we expect more on clang support in the Tegra platform port?
Or should we tune the doc. to state clang is not supported on all platform ports in general?

webgeek1234, 2025-06-30:
@vwadekar Was anything discovered internally about why binaries built with clang don't boot on tegra?

webgeek1234, 2025-08-12:
@odeprez With no response from the Tegra maintainer in over three months, perhaps the docs should just be updated and this issue closed.

odeprez, 2025-08-25:
Hi @webgeek1234 would you be up for submitting this change?

webgeek1234, 2025-09-20:
The documentation change has been merged.

closed 2025-09-20 by webgeek1234