TrustedFirmware-A/trusted-firmware-a #121: CPU workaround for erratum was missing

zhangyiru, 2025-06-27:
hi,all
i am new to TF-A.i find missing erratum keywords in ATF log.
WARNING: BL31: cortex_x925: CPU workaround for erratum 3701747 was missing!

1.this patch add erratum 3701747 and define check_erratum_cortex_x925_3701747 in lib/cpus/aarch64/cortex_x925.S.
https://github.com/ARM-software/arm-trusted-firmware/commit/511148ef5077dfb8f6bc7b9655e4ac19e16c4af0
2.this patch change check_erratum_cortex_xxx to cpu_reset_prologue func
https://github.com/ARM-software/arm-trusted-firmware/commit/89dba82dfa85fea03e7b2f6ad6a90fcd0aecce55#diff-effba13ad91f874b59e43afa91340e00c744907535bc33cbbae4b02853bbb72b
but the errata_common.c code that call check_erratum_cortex_xxx do not changed.
i think it is the reason of missing. plz check it again.

yiru.zhang
thx~

zhangyiru, 2025-06-30:
> Hi zhangyiru,
> 
> Are you adding `ERRATA_X925_3701747=1` to your platform makefile or to your build options?
> 
> macro - "check_erratum_ls cortex_x925, ERRATUM(3701747), CPU_REV(0, 1)" will add the function `check_erratum_cortex_x925_3701747`
> 
> Ref - https://github.com/ARM-software/arm-trusted-firmware/blob/master/include/lib/cpus/aarch64/cpu_macros.S#L555

zhangyiru, 2025-06-30:
thanks for your reply.
i enable ERRATA_X925_3701747=1, but it reports a error, undefined symbol `check_erratum_cortex_x925_3701747'

lib/cpus/errata_common.c
check_erratum_cortex_x925_3701747

lib/cpus/aarch64/cortex_x925.S
#include <cpu_macros.S>
check_erratum_ls cortex_x925, ERRATUM(3701747), CPU_REV(0, 1)

include/lib/cpus/aarch64/cpu_macros.S
.macro check_erratum_ls _cpu:req, _cve:req, _id:req, _rev_num:req
	func_compat check_erratum_\_cpu\()_\_id
		cpu_rev_var_ls \_rev_num
		ret
	endfunc_compat check_erratum_\_cpu\()_\_id
.endm

zhangyiru, 2025-06-30:
sorry to close this issue accidentally..

zhangyiru, 2025-07-29:
hi @govindraj-arm ,
i find diff with cortex_x925.S and cortex_a725.S.

[image]

[image]

https://github.com/ARM-software/arm-trusted-firmware/commit/89dba82dfa85fea03e7b2f6ad6a90fcd0aecce55#diff-effba13ad91f874b59e43afa91340e00c744907535bc33cbbae4b02853bbb72bL24

so, i add the line in cortex_x925.S. 
.global check_erratum_cortex_x925_3701747
then this build fail is solved.
https://github.com/TrustedFirmware-A/trusted-firmware-a/issues/121#issuecomment-3017689688

maybe accidentally deleted?

thx,
yiru.zhang

govindraj-arm, 2025-07-29:
Hi @zhangyiru ,

Thanks for the details.

Strangely this still doesn't make sense to me, its not causing a problem when building for FVP platform, I see we have declaration in header `include/lib/cpus/aarch64/cortex_x925.h` and used in `lib/cpus/errata_common.c` file 
so when compiling it should be be still ok?

Also I see apart from FVP, tc3 and mtk-mt8196 platforms using x925.S file.
They all seem to compile without any problems in CI. Based on today's CI nightly results I can see they build
x925.S without any issues.

for tc3 - https://ci.trustedfirmware.org/job/tf-a-builder/298389/consoleFull
for mt8196 - https://ci.trustedfirmware.org/job/tf-a-builder/298379/console

Sorry I am failing to connect the dots why we need to have the assembly global export.

--
Thanks,
Govindraj

loosper-arm, 2025-07-30:
No, AFAIU, platforms don't use alternative cpu library implementations. This problem is only reproducible with CTX_INCLUDE_EL2_REGS=1 and we don't do that for neither platform in CI. @govindraj-arm it might be useful to expand the errata-all build config to have an SPM too (or maybe merge with the all config).

Anyway, here's the fix: https://review.trustedfirmware.org/c/TF-A/trusted-firmware-a/+/41365

govindraj-arm, 2025-07-30:
Thanks Boyan,

I have merged the Patch, will look into CI changes.