TrustedFirmware-A/trusted-firmware-a #151: tegra: Tegra210 fails to boot

webgeek1234, 2025-09-20:
I recently began transferring my project to using v2.12 for t210, t186, and t194. And my tests with t210 failed to boot. On both the l4t r32.7.6 boot stack and the android rel-shield-r one. So I began stepping backwards to through release tags to find the last working tag for the arch. And found that it was v2.7. Three years ago.

Next I started bisecting to find where the breakage occurred. 0fe7b9f2bcdf754c483399c841e5f0ec71e53ef3 broke the arch during the v2.8 development cycle. Then 40a4e2d84c38ffae899eaa2c33c1e280312919cf fixed that during the v2.9 dev cycle. However, f8a35797b919d8ea041480bd5eb2a334e7056e0b quickly broke things again. cb6c8efc4f2dd951b2b51801a12ba1755d49fe79 was supposed to fix that, but it actually doesn't. I get send_smc errors spammed in the boot log and boot never completes. However, that doesn't even matter because before that was merged, 0b9f05fcaea069bff6894d99ec5babc4be29ca67 landed which prevents t210 from even loading tf-a enough to start logging. End result: v2.9 doesn't boot for two separate reasons. And nothing changed from then until now.

If I revert cb6c8efc4f2dd951b2b51801a12ba1755d49fe79, 0b9f05fcaea069bff6894d99ec5babc4be29ca67, and f8a35797b919d8ea041480bd5eb2a334e7056e0b on top of v2.12, t210 boots as expected. This is not a valid solution however, because it reverts an intentional change to meet spec and it also breaks the other tegra platforms.
<br/>
This all raises a big question for me. Are the tegra platforms maintained at all? I looked at using tf-a release tags, expecting that the platforms were being verified and confirmed stable. Tegra210, Tegra186, and Tegra194 are listed in docs as supported. However:
a) Tegra210 hasn't even booted for three years.
b) Tegra186 throws mmu faults galore, causing display corruption during boot and likely other issues. And the [maintainer said](https://review.trustedfirmware.org/c/TF-A/trusted-firmware-a/+/35535/1#message-e4cfa0537335d6d88de06352eab731eff5ea4eb7) that Tegra186 support was developed against an internal bootloader that was never released to the public, so it has never worked right. And since that platform is EOL, it is unlikely the closed source updates will ever be released.
c) Tegra194 seems to work okay, but with the condition of the other two archs, my confidence is waning.
d) As a related note, it appears that Nvidia has no intention of pushing support for the newer platforms such as Tegra234 and Tegra264 and intends to only support those in downstream forks with no commit history. Which makes it near impossible to unify back to a single repo for supporting multiple archs inside a project.

So my question is: Is there any intention to maintain tegra support in upstream tf-a? If not, then the illusion that it is needs to be removed. Mark the platforms as unmaintained, so any potential users know to expect to put in work to make things usable. Or remove the platforms entirely.

odeprez, 2025-12-18:
Hi, thanks for the detailed summary. As I understand it, the issue is very much Nvidia platform specific. It looks this should all be followed up through Nvidia forums as Varun suggested.

On the generic TF-A, I just merged https://review.trustedfirmware.org/c/TF-A/trusted-firmware-a/+/43279
Please submit additional changes as required, TF-A maintainers will support through the regular review process.

Thank you.

odeprez, 2025-12-19:
Hi,
> But my threads on the Nvidia forum were closed as 'not supported'.

Yes I saw this however not something trustedfirmware.org's TF-A project monitors or controls.

>  That appears to be a dead end for support. And even after 43279 was merged, Tegra210 still won't boot due to the fast smc issue I described in the issue description. 

Yes I understood it from your useful and detailed description, and this ticket captures it as a reference. I proposed a follow up with additional patches submitted to tforg's gerrit that TF-A reviewers will be happy to look at. As to whom write those patches I agree this is unknown.

>  That is a bug in the tf-a code base, though yes, it is Tegra specific. And I have no idea how to fix that myself which is partially why I opened this in the first place.

TF-A as generic community project mostly uses Arm's FVP for its validation + couple of common silicon platforms but doesn't own and test all platforms and flavors it supports (100+). For the latter the project fully relies on platform maintainers to review, approve and validate changes for said platforms. 

Regards,
Olivier.